NVIDIA/warp #645: [QUESTION]

JMLee0106, 2025-04-11:
Hi, first of all, thank you for the awesome library.

I have a question about whether it's possible to convert between 2D and 1D tiles.

Here’s an example:

```y_tile = wp.tile_zeros(shape=(n, 1), dtype=wp.float32)
wp.tile_matmul(B_tile, c_tile, y_tile)

L_tile = wp.tile_cholesky(A_tile)
x_tile = wp.tile_cholesky_solve(L_tile, y_tile)
```

To use `tile_matmul`, I need `y_tile` to be a 2D tile. However, it seems that `tile_cholesky_solve` only accepts 1D tiles as input. To make this work, I think I need to convert a 2D tile to a 1D tile.

Is there a way to perform this conversion? (And vice versa: 1D tile to 2D tile?)

Thanks in advance!

daedalus5, 2025-04-11:
Thanks for your question. We don't have a way to do this yet, but it can be added. Also, we should probably add a `tile_matmul()` variant that does matrix-vector multiplication. In the meantime, you can break the problem down like this:

```python
import warp as wp
import numpy as np

BLOCK_DIM = 32
TILE = 8

@wp.kernel
def kernel_matmul(
    B: wp.array2d(dtype=float),
    C: wp.array2d(dtype=float),
    Y: wp.array2d(dtype=float),
):
    i, j, = wp.tid()

    b = wp.tile_load(B, shape=(TILE, TILE))
    c = wp.tile_load(C, shape=(TILE, 1))

    y_2d = wp.tile_matmul(b, c) # (TILE, 1)

    wp.tile_store(Y, y_2d)

@wp.kernel
def kernel_cholesky(
    A: wp.array2d(dtype=float),
    Y: wp.array1d(dtype=float),
    X: wp.array1d(dtype=float)
):
    i, j = wp.tid()

    a = wp.tile_load(A, shape=(TILE, TILE))
    l = wp.tile_cholesky(a)
    y_1d = wp.tile_load(Y, shape=TILE)

    x = wp.tile_cholesky_solve(l, y_1d)
    wp.tile_store(X, x)

wp.set_device("cuda:0")

A_h = np.ones((TILE, TILE), dtype=float) + 5 * np.diag(np.ones(TILE), 0)
B_h = np.ones((TILE, TILE), dtype=float)
C_h = np.ones((TILE, 1), dtype=float)

A_wp = wp.array2d(A_h, dtype=float)
B_wp = wp.array2d(B_h, dtype=float)
C_wp = wp.array2d(C_h, dtype=float)

X_wp = wp.zeros(TILE, dtype=float)
Y_wp = wp.zeros((TILE, 1), dtype=float)

wp.launch_tiled(kernel_matmul, dim=[1, 1], inputs=[B_wp, C_wp, Y_wp], block_dim=BLOCK_DIM)
wp.launch_tiled(kernel_cholesky, dim=[1, 1], inputs=[A_wp, Y_wp.reshape((TILE,)), X_wp], block_dim=BLOCK_DIM)

print(X_wp.numpy())
```

JMLee0106, 2025-04-12:
I understand your point. Thank you for your answer!

closed 2025-04-12 by JMLee0106

daedalus5, 2025-04-25:
FYI: we've added support for this sort of conversion in commit [c742435](https://github.com/NVIDIA/warp/commit/c742435d2338dbc68b606dfb347c8931759e88a3)

JMLee0106, 2025-04-25:
Awesome! This will be really useful. Thank you!