NVIDIA/warp #647: [QUESTION] Support for Plate/Shell Elements or Bending in `warp.fem`

akrivx, 2025-04-11:
Hi,

I'm exploring how to implement bending forces in `warp.fem` and was wondering:

1. Is there any support for plate or shell elements (e.g. Reissner-Mindlin or Kirchhoff-Love)?
2. If not, is there any recommended way to implement bending behaviour over triangle meshes using `warp.fem`?
3. Are there any existing examples demonstrating bending forces, e.g., via dihedral angles or curvature-based models?

I'd be very interested in knowing whether a native warp.fem approach is possible or planned, particularly for thin-shell or cloth-like simulations.

Thanks,
Chris

gdaviet, 2025-04-14:
There's no support for second-order derivatives at the moment; this is definitely on my roadmap, but probably a few months away (I'd currently aim for 1.9.0).
Assuming no in-element bending (which should hold for triangular meshes), it should still be possible to implement a Discrete Shell kind of dihedral angle energy, though I've not tried yet. I would imagine integrating on interior sides something similar to

```python
@fem.integrand
def dihedral_angle(s: fem.Sample, sides: fem.Domain):
  cells = fem.cells(sides, s)
  s0 = fem.to_inner_cell(s)
  s1 = fem.to_outer_cell(s)
  n0 = fem.normal(cells, s0)
  n1 = fem.normal(cells, s1)
  angle = wp.atan2(wp.length(wp.cross(n0, n1)), wp.dot(n0, n1))

  L = fem.measure(sides, s)   # edge length
  h = L / fem. measure(cells, s0) + L /  fem. measure(cells, s1)   #sum of tri heights
  return angle / h
```

Deriving the corresponding linearized forms is left as an exercise :) (but curious to know how it goes)

gdaviet, 2025-04-15:
Yes, you can make this assumption as long as you're sticking with linear (P1) shape functions (for higher degrees the nodes will be arranged such that the vertex nodes still come first, then edge nodes, then interior nodes)

akrivx, 2025-04-15:
Excellent, thank you.