NVIDIA/warp #649: [QUESTION] How to generate .ptx without launch it?

ChenlizheMe, 2025-04-12:
Hey guys, I'm trying to compile all of Warp's kernels into PTX and then use them in C++. However, I noticed that it seems necessary to `launch` a kernel in order for it to be compiled. This becomes very inconvenient when a large number of kernels need to be compiled. Is there a way to compile them without launching the kernels?

shi-eric, 2025-04-12:
You can use wp.load_module(device) to force compilation of the module’s kernels without launching them one-by-one. https://github.com/NVIDIA/warp/blob/f4572ba414fd60893e08bc46fbf3df51f0846661/warp/context.py#L6078

ChenlizheMe, 2025-04-12:
> You can use wp.load_module(device) to force compilation of the module’s kernels without launching them one-by-one.
> 
> [warp/warp/context.py](https://github.com/NVIDIA/warp/blob/f4572ba414fd60893e08bc46fbf3df51f0846661/warp/context.py#L6078)
> 
> Line 6078 in [f4572ba](/NVIDIA/warp/commit/f4572ba414fd60893e08bc46fbf3df51f0846661)
> 
>  def load_module(

It works! Thx a lot!

closed 2025-04-12 by ChenlizheMe