NVIDIA/warp #653: [QUESTION] Copy from host-to-device inside a cuda graph

CliffieVanR, 2025-04-15:
Hi, I have a long serial graph that I'm capturing, with the last intended operation being a copy from host to device via dst = src.numpy(). However, when that code is executed, I get the following error:

Warp CUDA error 906: operation would make the legacy stream depend on a capturing blocking stream (in function cuda_context_set_stream, C:\dev\warp\warp\native\warp.cu:2151)
Warp CUDA error 906: operation would make the legacy stream depend on a capturing blocking stream (in function memcpy_d2h, C:\dev\warp\warp\native\warp.cu:754)

I've tried switching execution of everything to a non-default/legacy stream, to no avail.

I've tried also removing the copy outside of the graph capture and instead performing it after executing the graph, but this just returns garbage.

Is there any way to do this?

Thanks in advance!

shi-eric, 2025-04-16:
Right, calling `numpy()` on an array allocated on the GPU will always use the legacy default stream in CUDA and can't be used as part of a graph capture. One solution would be to allocate a Warp array on the CPU with `pinned=True` (assuming there is enough pinned memory available) and then use `wp.copy()` inside the stream capture operation: https://nvidia.github.io/warp/modules/concurrency.html#array-copying

Doing the `wp.copy()` outside the graph can also be made to work. I'm not sure why you were getting garbage. If you have a simple repro, we can offer more specific advice.

CliffieVanR, 2025-04-16:
Hi Eric,

That's good to know. I probably ought to look into why I'm getting garbage
outside the graph - quite possibly an error on my end. But it sounds like
using the pinned memory would be a perfectly decent solution. Many thanks!

Cliff

closed 2025-05-07 by shi-eric