NVIDIA/warp #665: Investigate `test_qr_eigenvalues` failure on Blackwell

shi-eric, 2025-04-19:
This seems to happen only on the Blackwell GitLab runner I set up.

```
======================================================================
test_qr_eigenvalues_cuda_0 (warp.tests.test_fem.TestFemUtilities)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/builds/omniverse/warp/warp/tests/unittest_utils.py", line 340, in test_func_wrapper
    test_func(test, device)
  File "/builds/omniverse/warp/warp/tests/unittest_utils.py", line 323, in test_func
    wp.launch(kernel, dim=dim, inputs=args, device=device)
  File "/builds/omniverse/warp/warp/tests/unittest_utils.py", line 234, in __exit__
    self.test.fail(f"Unexpected output:\n'{s.rstrip()}'")
AssertionError: Unexpected output:
'Module warp.tests.test_fem 84892fd load on device 'cuda:0' took 6.49 ms  (cached)
Error, expect_near() failed with tolerance 1e-08
	 Expected: 2
	 Actual: 2
Error, expect_near() failed with tolerance 1e-08
	 Expected: 2.5
	 Actual: 2.5'
----------------------------------------------------------------------
```

shi-eric, 2025-04-19:
```
======================================================================
FAIL: test_qr_eigenvalues_cuda_0 (__main__.TestFemUtilities.test_qr_eigenvalues_cuda_0)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/horde/code-projects/warp/warp/tests/unittest_utils.py", line 340, in test_func_wrapper
    test_func(test, device)
  File "/home/horde/code-projects/warp/warp/tests/unittest_utils.py", line 322, in test_func
    with CheckOutput(self):
         ^^^^^^^^^^^^^^^^^
  File "/home/horde/code-projects/warp/warp/tests/unittest_utils.py", line 234, in __exit__
    self.test.fail(f"Unexpected output:\n'{s.rstrip()}'")
AssertionError: Unexpected output:
'Module __main__ 84892fd load on device 'cuda:0' took 60477.31 ms  (compiled)
Error, expect_near() failed with tolerance 1e-08
    Expected: 2
    Actual: 2
    Absolute difference: 2.38419e-07
Error, expect_near() failed with tolerance 1e-08
    Expected: 2.5
    Actual: 2.5
    Absolute difference: 2.38419e-07'

----------------------------------------------------------------------
```

@gdaviet Any issue with just bumping up the tolerance?

gdaviet, 2025-04-19:
> 
> [@gdaviet](https://github.com/gdaviet) Any issue with just bumping up the tolerance?

Sounds reasonable to me!

closed 2025-04-20 by shi-eric