NVIDIA/warp #669: [QUESTION]

ChenlizheMe, 2025-04-21:
Hello, guys

I've noticed an inconsistency in CUDA version information when compiling the same kernel code. Sometimes the generated PTX shows:
```
Cuda compilation tools, release 12.8, V12.8.61
```
while other times it shows:
```
Cuda compilation tools, release 12.1, V12.1.55
```
However, when I run `nvcc -V` on my computer, I get:
```
Cuda compilation tools, release 12.1, V12.1.66
```

Questions:
1. Why am I seeing different CUDA toolchain versions in the PTX output?
2. How can I consistently control which nvcc version is used for PTX generation?
3. Is there a way to force the compilation to use exactly the version shown by `nvcc -V`?

Any suggestions for debugging or resolving this would be appreciated!

shi-eric, 2025-04-21:
> Why am I seeing different CUDA toolchain versions in the PTX output?

You might been running some programs using a pre-built Warp from PyPI, which comes statically linked with libraries from the CUDA 12.8 toolkit and other programs using a version of Warp you compiled from source, which comes statically linked with libraries from the CUDA 12.1 toolkit on your computer. In contrast to some other CUDA Python libraries, we statically link in things like NVRTC rather than having it dynamically loaded from your environment at runtime.

> How can I consistently control which nvcc version is used for PTX generation?

Just keep track of which version of Warp you're running your programs with. The CUDA toolkit version that's statically linked into the Warp library will be printed with the `wp.init()` output.

> Is there a way to force the compilation to use exactly the version shown by nvcc -V?

Build Warp from source.

ChenlizheMe, 2025-04-23:
That's helpful, thanks a lot!

closed 2025-04-23 by ChenlizheMe