NVIDIA/warp #671: [QUESTION] Clarification on surface FEM and rigid body motion under out-of-plane loading

akrivx, 2025-04-22:
Hi,

I’ve been using `warp.fem` for cloth simulation, utilising a corotational SVD-based model with Green strain over 3D triangle meshes.

While investigating rigid body behaviour, I noticed that the cloth rotates properly when external forces (e.g., gravity) act in-plane (e.g., along X or Y), but not when they act perpendicular to the surface (Z direction). Initially I suspected that `fem.grad` might suppress out-of-plane variation, but after inspecting gradients for tilted triangles, I confirmed that the test and trial gradients do include non-zero components in the normal direction.

Despite this, the cloth still does not rotate under Z-directed gravity, even in asymmetric setups. I’ve also tried adding bending terms (e.g., discrete shell-style energies), but the issue persists — suggesting that the problem lies deeper.

Are there any known limitations or subtleties in Warp’s surface FEM implementation — such as the way test/trial functions are defined, or integration weights are computed — that could suppress rigid-body torque under out-of-plane forces?

Thanks,
Chris

gdaviet, 2025-04-22:
Hi Chris,

More context about the setup would be helpful; I don't see a priori what would prevent warp.fem from representing out-of-plane forces and torques (see a minimal out-of-plane elasticity below). 
Maybe have a look at the linear system conditioning; if the elastic term is extremely stiff that could damp out the rotational motion? In that case Mixed FEM formulations could help

```python
import warp as wp
import warp.fem as fem
import warp.examples.fem.utils as fem_example_utils

import numpy as np

@wp.func
def hooke_stress(S: wp.mat33, lame: wp.vec2):
    strain = S - wp.identity(n=3, dtype=float)
    return 2.0 * lame[1] * strain + lame[0] * wp.trace(strain) * wp.identity(
        n=3, dtype=float
    )

@wp.func
def hooke_hessian(tau: wp.mat33, sig: wp.mat33, lame: wp.vec2):
    return wp.ddot(hooke_stress(sig + wp.identity(n=3, dtype=float), lame), tau)

def deform_along_z(positions, z_scale=1.0):
    pos = positions.numpy()
    pos_z = z_scale * np.cos(3.0 * pos[:, 0]) * np.sin(4.0 * pos[:, 1])
    pos = np.hstack((pos, np.expand_dims(pos_z, axis=1)))
    return wp.array(pos, dtype=wp.vec3)

resolution = 5

positions, tri_vidx = fem_example_utils.gen_trimesh(res=wp.vec2i(resolution))

positions = deform_along_z(positions, z_scale = 0.0)
geo = fem.Trimesh3D(tri_vertex_indices=tri_vidx, positions=positions)

u_space = fem.make_polynomial_space(geo, dtype=wp.vec3)
u_trial = fem.make_trial(u_space)
u_test = fem.make_test(u_space)

# LHS and rhs
@fem.integrand
def lhs(s:fem.Sample, u: fem.Field, v:fem.Field):
    # mass
    M = wp.dot(u(s), v(s)) 

    # membrane stiffness
    du = fem.grad(u, s)
    dv = fem.grad(v, s)
    K = hooke_hessian(du, dv, wp.vec2(1.0))

    return M + K

@fem.integrand
def rhs(s:fem.Sample, v:fem.Field):
    # gravity 
    return wp.dot(wp.vec3(0.0, 0.0, -10.0), v(s))

A = fem.integrate(lhs, fields={"u":u_trial, "v": u_test}, output_dtype=float)
b = fem.integrate(rhs, fields={"v": u_test},  output_dtype=wp.vec3)

# Dirichlet BC projector

@fem.integrand
def bc(domain: fem.Domain, s:fem.Sample, u: fem.Field, v:fem.Field):
    if domain(s)[0] <= 0.2:
        return wp.dot(u(s), v(s))
    return 0.0

proj = fem.integrate(bc, fields={"u":u_trial, "v": u_test}, output_dtype=float, assembly="nodal")
fem.project_linear_system(A, b, proj)

# solve linear system
du = fem.make_discrete_field(u_space)
fem_example_utils.bsr_cg(A=A, b=b, x=du.dof_values)

# plot resulting displacement field
renderer = fem_example_utils.Plot()
renderer.add_field("solution", du)
renderer.plot(options={"solution": {"displacement": {}}})
```

akrivx, 2025-04-23:
Hi,

Thank you for the earlier example. I’ve now built a minimal reproducible example to clarify further the issue I’m encountering.

The code below adapts your linear elasticity example to implement a nonlinear corotational elasticity model, using SVD-based rotation removal — essentially the same formulation you shared previously [here](https://github.com/NVIDIA/warp/issues/589#issuecomment-2742639582) (but I get the same results with a slightly more complex corotational elasticity model I've implemented).

Everything is set up using `warp.fem`. The mesh lies in the XY plane, and gravity acts in Z. I use implicit dynamics solved via Newton-Raphson iterations with mass and stiffness contributions, and have verified that the deformation gradients contain non-zero Z components when the mesh deforms.

However, the cloth does not respond at all to out-of-plane loading. Even rigid-body behaviour like falling or rotating (with one point or edge pinned) doesn’t occur — the system simply converges with a near-zero update.

This differs from both:
- Your original linear example, which shows visible Z-displacement under gravity
- The same model running on volumetric hexahedral meshes, where full 3D motion works correctly

It could be a numerical issue, as you suggested. I just wanted to verify this is not an inherent limitation of `warp.fem` that would filter out Z-force coupling.

The example includes a simple toggle to step the simulation frame-by-frame and a readout of mean displacement in Z and total energy. You can left-click anywhere in the Polyscope window to start/stop the simulation.

```python
import numpy as np
import warp as wp
import warp.sparse as sp
import warp.utils as utils
import warp.fem as fem
import warp.fem.linalg as linalg
import warp.examples.fem.utils as fem_example_utils
import polyscope as ps
import polyscope.imgui as psim

@wp.func
def symmetric_strain(F: wp.mat33):
	U = wp.mat33()
	sig = wp.vec3()
	V = wp.mat33()
	wp.svd3(F, U, sig, V)
	S = V * wp.diag(sig) * wp.transpose(V)
	return S

@wp.func
def symmetric_strain_delta(F: wp.mat33, dF: wp.mat33):
	# see supplementary of `WRAPD: Weighted Rotation-aware ADMM`, Brown and Narain 21
	U = wp.mat33()
	sig = wp.vec3()
	V = wp.mat33()
	wp.svd3(F, U, sig, V)

	Ut = wp.transpose(U)
	Vt = wp.transpose(V)
	
	dF_loc = Ut * dF * V
	SigdF_loc = wp.diag(sig) * dF_loc
	
	sig_op = wp.mat33(wp.vec3(sig[0]), wp.vec3(sig[1]), wp.vec3(sig[2]))
	dSig = wp.cw_div(SigdF_loc + wp.transpose(SigdF_loc), sig_op + wp.transpose(sig_op))
	dS = V * dSig * Vt
	
	return dS

@fem.integrand
def defgrad(u: fem.Field, s: fem.Sample):
	return fem.grad(u, s) + wp.identity(n=3, dtype=float)

@wp.func
def hooke_energy(S: wp.mat33, lame: wp.vec2):
	strain = S - wp.identity(n=3, dtype=float)
	return 0.5 * wp.ddot(strain, hooke_stress(S, lame))

@wp.func
def hooke_stress(S: wp.mat33, lame: wp.vec2):
	strain = S - wp.identity(n=3, dtype=float)
[... 159 lines cut ...]
	if simulate:
		advance(u_space, domain, u_cur, v_cur, u_test, u_trial, proj, DT, DENSITY, LAME, FORCE)

		u = u_cur.dof_values.numpy()
		x = rest_pos + u

		ps_mesh.update_vertex_positions(x)
		ps_mesh.add_scalar_quantity('displacement', np.linalg.norm(u, axis=1), cmap='coolwarm', enabled=True)

		E_t = fem.integrate(
			total_energy,
			domain=domain,
			fields={'u_cur': u_cur, 'v_cur': v_cur},
			values={'density': DENSITY, 'force': FORCE, 'lame': LAME}
		)

		psim.TextUnformatted(f'Energy: {E_t:.3e}')
		psim.TextUnformatted(f'Mean uz: {u[:, 2].mean():.3e}')

ps.set_user_callback(simulate_callback)
ps.show()
```

Thanks,
Chris

akrivx, 2025-04-23:
That's excellent, thank you. I'm glad it's not a `warp.fem` limitation.

If I get it right, energy is now a function of the stretch tensor $S$ (or whatever it's called, computed from $F=R \cdot S$) instead of the singular values alone.

As a result, the variation of the energy density function $\psi$ now depends on the variation of $S$, $\delta S$, which is:

$$
\delta S = R^T \cdot \delta F + dR^T \cdot F
$$

Thus the various `wp.transpose(R) * fem.grad(x, s)` terms in your example above, where $dR$ is ignored.

gdaviet, 2025-04-24:
The previous implementation was already expressing the energy on `S` (but computing the exact derivative using `symmetric_strain_delta`). The issue was that the `Id` part of `F` was introducing a shearing mode as soon as the triangles would start rotating. 
To keep using the 3d energy, we need instead to complete the rank-2 F from the triangle deformation with an orthogonal rank-1 tensor; the second variant uses the SVD to do this (but that  complicates computing the exact derivative, so I did not bother doing this above). I believe another option would be to write F as  `grad(u) + Id - wp.outer(n, n) + wp.outer(def(n), def(n))`, where `def(n)`  is the deformed triangle normal, but again that complicates the derivative.

In the end, it might be simpler to use a 2D energy instead, rather than explicitly constructing a `1` singular value.

akrivx, 2025-04-24:
Very interesting, thanks. I suppose that by 2D energy, you mean a model in which we project the deformation gradient onto the tangent plane of each triangle (e.g. something like $F = P^T \nabla u P + I$, where $P$ is the $3 \times 2$ tangent basis) and evaluate a Green strain-based energy and its variations in 2D, right?

akrivx, 2025-04-24:
I'm seeing interesting results with a corotational Green strain model that uses the deformation gradient appearing in the previously indicated Zhai et al. publication, tailored to `warp.fem`'s weak form framework.

The key idea is to compute the deformation gradient $F \in R^{3 \times 2}$ from the transformation between the undeformed and deformed tangent bases:

$$
F = T_{def} \cdot R_{rest}^{-1}
$$

where $T_{def}$ and $T_{rest}$ are $3 \times 2$ matrices constructed from vertex positions in the current (deformed) and rest configurations, and $Q_{rest}$ is the upper-triangular matrix from a $QR$ factorisation of $T_{rest}$.

This naturally gives rise to the Green strain tensor:

$$
E = \frac{1}{2} (F^T F - I)
$$

and a 2D energy density:

$$
\psi(E) = 2 \mu E : E + \lambda (tr E)^2
$$

From this, both the internal force and tangent stiffness matrix are computed analytically and fully within the weak form, avoiding SVD or polar decomposition.

Happy to share a cleaned-up code example if you're interested.