NVIDIA/warp #683: [QUESTION] How to convert the data type of a tile?

chaoming0625, 2025-04-27:
Here is what I want to do. But I do not know what's wrong with this kernel.

```python

import warp
import numpy as np

@warp.kernel
def kernel(
    a: warp.array(dtype=bool),
    b: warp.array(dtype=float),
):
    data = warp.tile_load(a, TILE_SIZE)
    data0 = warp.untile(data)
    data1 = warp.where(data0, 1., 0.)
    data2 = warp.tile(data1)
    warp.tile_store(b, data2)

m = 128
TILE_SIZE = m

data_a = warp.array(np.random.rand(m) < 0.3, dtype=bool)
data_b = warp.array(np.zeros(m), dtype=float)

warp.launch_tiled(kernel, dim=[1], inputs=[data_a, data_b], block_dim=128)

print(data_b.numpy())

```

chaoming0625, 2025-04-27:
I also tried to use `warp.tile_map` by the following syntax. But it does not work. 

```python

@warp.func
def where(s):
  return warp.where(s, 1., 0.)

warp.tile_map(where, data)

```

This produce the error of inconsistent data type in `warp.tile_map` due to the mismatch of `bool` and `float`.

daedalus5, 2025-05-01:
Would this work?

```python
import warp
import numpy as np

TILE_SIZE = 128

@warp.kernel
def kernel(
    a: warp.array(dtype=int),
    b: warp.array(dtype=float),
):
    data = warp.tile_load(a, TILE_SIZE)
    data0 = warp.untile(data)
    data1 = warp.where(data0, 1., 0.)
    data2 = warp.tile(data1)
    warp.tile_store(b, data2)

data_a = warp.array(np.random.rand(TILE_SIZE) < 0.3, dtype=int)
data_b = warp.array(np.zeros(TILE_SIZE), dtype=float)

warp.launch_tiled(kernel, dim=[1], inputs=[data_a, data_b], block_dim=128)

print(data_a.numpy())
print(data_b.numpy())
```

chaoming0625, 2025-05-05:
Your example is converting `int` to `float`. However, my example tried to convert `bool` to `float`. Is it possible to provide a general interface such as `tile_asarray` to convert the tiled data type?

daedalus5, 2025-05-05:
Yep, we'll add this to our list.

chaoming0625, 2025-05-09:
Is it possible to provide such support as soon as possible? This is an urgent requirement for me. Thanks.  :heart_eyes:

chaoming0625, 2025-05-09:
@shi-eric @daedalus5

daedalus5, 2025-05-09:
Sure, I can work on this next.

daedalus5, 2025-05-13:
Hi @chaoming0625 , we added `wp.tile_astype()` to Warp in this commit: [a026979](https://github.com/NVIDIA/warp/commit/a02697972fdd19f42764db3bf139bf05632352ad)

closed 2025-05-13 by daedalus5