NVIDIA/warp #684: [REQ] `tile_load` or `tile_atomic_add` at given index

chaoming0625, 2025-04-28:
Here is what I want to do:

```python
@warp.kernel
def kernel(
    index: warp.array2d(dtype=int),
    weight: warp.array2d(dtype=float),
    out: warp.array1d(dtype=int),
):
    i = warp.tid()

    ind = warp.tile_load(index[i], n_conn)
    w = warp.tile_load(weight[i], n_conn)
    warp.tile_store(out, ind, w)

n_pre = 1000
n_post = 1000
n_conn = 20

index_ = warp.array(np.random.randint(0, n_post, (n_pre, n_conn)))
weights_ = warp.array(np.random.randn(n_pre, n_conn))
out = warp.empty(n_post, dtype=float)

warp.launch_tiled(kernel, inputs=[index_, weights_], outputs=[out], dim=n_pre, block_dim=32)

print(out.numpy())

```

The error message is:

```bash
warp.codegen.WarpCodegenError: Error while parsing function "kernel" :
    warp.tile_store(out, ind, w)
;Couldn't find function overload for 'tile_store' that matched inputs with types: 

```

This is because the current `tile_store` only support store a tile at a continuous positions, rather than the given index.

chaoming0625, 2025-04-29:
I am trying another syntax:

```python

    @warp.kernel
    def kernel(
        index: warp.array2d(dtype=int),
        weight: warp.array2d(dtype=float),
        out: warp.array1d(dtype=float),
    ):
        i = warp.tid()
        ind = warp.tile_load(index[i], n_conn)
        w = warp.tile_load(weight[i], n_conn)
        ind = warp.untile(ind)
        w = warp.untile(w)
        warp.atomic_add(out, ind, w)

    n_pre = 1000
    n_post = 1000
    n_conn = 40

    index_ = warp.array(np.random.randint(0, n_post, (n_pre, n_conn)))
    weights_ = warp.array(np.random.randn(n_pre, n_conn), dtype=float)
    out = warp.empty(n_post, dtype=float)

    # warp.launch(kernel, inputs=[index_, weights_], outputs=[out], dim=[n_pre, n_conn])
    warp.launch(kernel, inputs=[index_, weights_], outputs=[out], dim=[n_pre], block_dim=32)

    print(out.numpy())

```

This produces the error:

```
ValueError: Error while parsing function "kernel" at warp_exp1.py:71:
    ind = warp.untile(ind)
;untile() argument last dimension 40 does not match the expected block width 32

```

chaoming0625, 2025-04-30:
Similarly, we also need support for `tile_load` at the given index, rather than continuous positions. For example:

```python

index = [0, 3, 5]
warp.tile_load(a, index)

```

daedalus5, 2025-05-01:
This sort of problem isn't well suited for the Tile API I think. Could you use a SIMT approach instead?

```python
n_pre = 1000
n_post = 1000
n_conn = 20

@wp.kernel
def kernel(
    index: wp.array2d(dtype=int),
    weight: wp.array2d(dtype=float),
    out: wp.array1d(dtype=float),
):
    i, j = wp.tid()

    idx = index[i, j]
    w = weight[i, j]
    wp.atomic_add(out, idx, w)

index_ = wp.array(np.random.randint(0, n_post, (n_pre, n_conn)), dtype=int)
weights_ = wp.array(np.random.randn(n_pre, n_conn), dtype=float)
out = wp.empty(n_post, dtype=float)

wp.launch(kernel, dim=[n_pre, n_conn], inputs=[index_, weights_], outputs=[out])

print(out.numpy())
```

daedalus5, 2025-05-05:
I see, yeah. This is actually on our roadmap, so we'll be adding indexed load/stores.

closed 2025-09-05 by shi-eric