NVIDIA/warp #688: [BUG] Tile operations

thowell, 2025-04-29:
### Bug Description

The following code with tile operations produces unexpected results including NaNs with Warp nightly build 1.8.0.dev20250429.

```
import numpy as np
import warp as wp

def tile_cholesky(
  M: wp.array(dtype=wp.float32, ndim=2),
  L: wp.array(dtype=wp.float32, ndim=2),
  tilesize: int,
):
  @wp.kernel
  def cholesky(
    M: wp.array(dtype=wp.float32, ndim=2), L: wp.array(dtype=wp.float32, ndim=2)
  ):
    M_tile = wp.tile_load(M, shape=(tilesize, tilesize))
    L_tile = wp.tile_cholesky(M_tile)
    wp.tile_store(L, L_tile)

  wp.launch_tiled(cholesky, dim=(1,), inputs=[M, L], block_dim=32)

dim = 5
Mnp = np.eye(dim) * 2.0
Lnp = np.linalg.cholesky(Mnp)

Mwp = wp.array(Mnp, dtype=wp.float32)
Lwp = wp.zeros((dim, dim), dtype=wp.float32)

tilesize = 4
tile_cholesky(Mwp, Lwp, tilesize)

err = Lnp[:tilesize, :tilesize] - Lwp.numpy()[:tilesize, :tilesize]

print(f"Lnp - Lwp:\n{err}")
print(f"error(Lnp - Lwp):{np.linalg.norm(err)}")
```

@mmacklin @daedalus5

daedalus5, 2025-04-29:
Seems okay if you increase `tilesize` to match `dim`?

erikfrey, 2025-04-29:
Hi @daedalus5 - I think that's deliberate.  We would like to do cholesky on tiles smaller than the full array size.

daedalus5, 2025-04-29:
I see, yeah we want this to still work on the submatrix. Might be an issue with memory layout in `tile_load()`. Will look into this.

daedalus5, 2025-04-30:
This should be fixed with [ccb4db5](https://github.com/NVIDIA/warp/commit/ccb4db5666ed13494d0808fec0316b9466dc4192)

closed 2025-04-30 by daedalus5