NVIDIA/warp #690: [QUESTION] Syntax in VDB's evaluate_self_contact_force_norm

CliffieVanR, 2025-04-30:
Hi, I wanted to check if the syntax here in evaluate_self_contact_force_norm:

    # C2 continuity calculation
    tau = collision_radius * 0.5
    if tau > dis > 1e-5:

I'm not too familiar with Python, but a colleague tells me this may be a check for 'dis' being within the interval tau and 1e-5. Is that correct?

shi-eric, 2025-04-30:
Yes, it's called a chained comparison in Python and is equivalent to `tau > dis and dis > 1e-5`.

CliffieVanR, 2025-04-30:
Thanks for confirming!

CliffieVanR, 2025-04-30:
Just so I'm clear on this, the semantics are that the collision force is
linear if the particle is very close to or far from the edge or tri it's
colliding with, otherwise it's using that tau^2/dist formulation? Is there
a justification for this written anywhere? I don't recall seeing it in the
VDB paper.

shi-eric, 2025-04-30:
> Just so I'm clear on this, the semantics are that the collision force is
> linear if the particle is very close to or far from the edge or tri it's
> colliding with, otherwise it's using that tau^2/dist [...]

I'll direct this to @AnkaChan as the expert here

AnkaChan, 2025-04-30:
@CliffieVanR You are right, the contact energy is a two-stage function:

[image]

The reason why we switch back to linear force when the contacting primitives are too close is to avoid dividing by zero.

CliffieVanR, 2025-05-01:
Hi Anka,

Many thanks. Can I ask which paper that equation comes from? I've been
working from your paper here Vertex Block Descent
<https://arxiv.org/pdf/2403.06321>. If it's there, apologies: I've missed
it. (Also, it does look a little different from what's actually implemented
in Warp's VBD integrator. May I ask why?

Speaking of dividing by zero, I found, when using the analytic grid caused
by add_cloth_grid, that in some cases I was seeing distances of precisely
zero being used to compute normals in both self-contact kernels. I wonder
if this might be caused by duplicate edges and/or vertices being produced
by the add_cloth_griid code in model.py? I've worked around it for now by
bailing if dis is not greater than zero.

closed 2025-11-06 by shi-eric