NVIDIA/warp #693: [QUESTION] Damping term in VDB

CliffieVanR, 2025-04-30:
Am I correct to think of the damping term in VDB, called 'spring_kd' in the code, is damping both the elastic spring forces *and* the inertial force?

        # compute damping
        k_d = tri_materials[tri_id, 2]
        h_d = h_tri * (k_d / dt)

        #f_d = h_d * (prev_pos[particle_index] - pos[particle_index])

prev_pos here is the position prior to the time-step integration, so increase sping_kd/k_d has the effect of slowing a piece of cloth's fall under gravity.

I'm having a hard time finding a damping term for my particular use case that prevents extreme spring oscillations without also causing an artificial resistance to gravity. I see in the paper for the cloth example you used 1e4 and 1e-5 for stiffness and damping respectively. I'm finding such values (used in correspondence with cloth density of 0.15 kgm-2) give me very peculiar results.

Thanks in advance!
Cliff

AnkaChan, 2025-04-30:
Hi Cliffie,

I assume you are talking about VBD (vertex block descent) instead of VDB.
First, the damping term for the elasticity is controlled by `tri_kd`, not `spring_kd`. Currently, the `spring_kd` parameter is not effective for VBDIntegrator since we have not implemented any spring energy yet.
Please see the arguments list for `add_cloth_mesh` and `add_cloth_grid` for reference.

Also, for cloth simulation, I would recommend using g/cm/s system to provide better results. It looks like you are using kg.

CliffieVanR, 2025-05-01:
Hi Anka,

Yes, you're right: I meant VBD, and I also meant tri_kd :).

I'd been thinking of evaluate_stvk_force_hessian as a spring-like
constraint. Is that wrong of me? At any rate, I'm still wondering if tri_kd
ought to be damping acceleration due to gravity?

Your recommendation to switch to g/cm/s: you're correct, we're currently
working in SI units. I see now that the Tau term in
evaluate_self_contact_force_norm is in units of length, so we need to be in
the same scale. Are there other reasons? Would working in g/cm provide more
stable numerical results? I've reported before that I've seen numerical
explosions with XPBD with a simple analytic grid (via add_cloth_grid) with
zero external forces and only spring constraints.

Thanks!
Cliff

CliffieVanR, 2025-05-01:
Apologies: I meant to say the 1e-5 constants in evaluate_self_contact_force_norm is in units of length and perhaps needs to be adjusted according to units? Tau is already derived directly from radius and is such is going to be in whatever units we're using.