NVIDIA/warp #730: [QUESTION] Model for VDB evaluate_dihedral_angle_based_bending_force_hessian

CliffieVanR, 2025-05-14:
Hi, can I ask which paper is used for the bending force implementation in VDB? So far as I can tell, the direction vectors of the forces don't correspond to those proscribed by Bridson et al.

Also, are you aware of any bugs in this function? I'm finding I get undesirable results when using it with garments (ie meshes with non-zero rest angles), although it seems to behave fine with the cloth grid.

Thanks in advance!
Cliff

AnkaChan, 2025-05-14:
Hi @CliffieVanR , the bending energy is k_e * (angle - rest_angle)^2.
Yes, we've noticed that currently the bending has some issues for non-zero rest angles.
Currently, you can walk around it by calling:
`self.model.edge_rest_angle.zero_()`
which will set all the rest angles to zero.

CliffieVanR, 2025-05-15:
Hi Anka,

Many thanks! Do you have a reference for the derivation of the force
directions and the hessian?

Best,
Cliff