NVIDIA/warp #736: [BUG] Featherstone with Free 6DOF joint results in incorrect simulation and gradients

dyumanaditya, 2025-05-16:
### Bug Description

Hello,

## My Setup
I'm trying to simulate the bounce example but with the ball being a rigid body sphere instead of a particle. This setup works well with the Semi-Implicit Euler integrator, and the gradients seem to be well conditioned but I run into several issues with the Featherstone integrator. The only difference in the setup is that I add a `free_joint` to the ball to be able to simulate it in generalized coordinates.

[image]

## Bug Description
The output states for both integrators differ with the featherstone states consisting of an angular rotation not present in the Euler state. There is no friction and no reason for the ball to incur any kind of angular velocity or change in orientation. In addition to this, the gradients differ significantly with the Featherstone gradients blowing up to an order of `10e25`

### Semi Implicit Euler final states after simulation
```
body_q  = tensor([1.0219, 1.9264, 0.0000, 0.0000, 0.0000, 0.0000, 1.0000])
body_qd = tensor([ 0.0000,  0.0000,  0.0000, -5.0018,  2.6256,  0.0000])
```

### Featherstone final states after simulation
```
joint_q  = tensor([1.0338, 1.9495, 0.0000, 0.0000, 0.0000, 0.5819, 0.8133])
joint_qd = tensor([ 0.0000,  0.0000,  3.4022,  1.6226, -0.7761,  0.0000])
``` 

### Semi Implicit Euler gradient of initial velocity state wrt loss

```
[-2.4142773  -0.31994405]
```

### Featherstone gradient of initial velocity state wrt loss
```
[-2.9803302e+25  1.2860158e+25]
```

## Attempts at Solving
I have observed that when I increase the default `update_mass_matrix_every=1` parameter of the Featherstone integrator, the gradients become smaller (but still very large), indicating a potential bug in the way the mass matrix is handled.

In case the following information is useful: If I comment out the #following lines in `integrator_featherstone.py` in the adjoint computation for the `dense_solve` function, the gradient seems to be more reasonable for the first iteration, but on multiple training iteration runs, the ball goes through the floor.

https://github.com/NVIDIA/warp/blob/23eb9ea711c5494ddeb2f51950a3d830a5864ed6/warp/sim/integrator_featherstone.py#L1428-L1430

I was also wondering if in the free-joint integration of jcalc_integrate, is it correct to compute the translation velocity as
https://github.com/NVIDIA/warp/blob/23eb9ea711c5494ddeb2f51950a3d830a5864ed6/warp/sim/integrator_featherstone.py#L604

Since `v_s` already represents the linear velocity of the joint-frame origin (and after quaternion rotation becomes the world-frame velocity of that same point), shouldn’t we simply use

```python
dpdt_s = v_s;
```
to avoid “sweeping” the origin around as if it were offset from the center of rotation?

Any help on this would be great,
Thanks!

shi-eric, 2025-05-16:
Hi @dyumanaditya, I truly appreciate the level of detail in this bug report and I would point others to it as a gold standard for how to file issues, but I don't expect that we will have time to look into this issue for a while. @momo-van posted an announcement yesterday about the in-progress migration of `warp.sim` into https://github.com/newton-physics/newton here https://github.com/NVIDIA/warp/discussions/735 The good news is that you should be able to file this issue in the Newton repository, where a fix would be delivered if one is identified.

etaoxing, 2025-05-16:
FYI this might be helpful, here's how I usually create a free-joint rigid object for Featherstone in [Rewarped](https://github.com/rewarped/rewarped):
```python
    def create_cube(self, builder):
        p = wp.vec3(0.5, 0.0, 0.055)
        q = wp.quat_identity()
        object_start_pose = wp.transform(p, q)

        box_size = dict(
            hx=0.026,
            hy=0.026,
            hz=0.026,
        )

        builder.add_articulation()
        b = builder.add_body(name="body_object")
        builder.add_shape_box(
            body=b,
            **box_size,
            mu=0.5,
            restitution=0.0,
            is_solid=True,
            has_ground_collision=True,
            has_shape_collision=True,
        )

        M = len(builder.joint_q)
        builder.add_joint_free(parent=-1, child=b, name="joint_object")
        builder.joint_q[M : M + 3] = object_start_pose.p
        builder.joint_q[M + 3 : M + 7] = object_start_pose.q
```

In particular, use `joint_q` to hold global coordinates, and don't add the xform to `body_q`. Might work for your case or in place of particles like in [`rewarped/envs/warp_examples/bounce.py`](https://github.com/rewarped/rewarped/blob/main/rewarped/envs/warp_examples/bounce.py)

dyumanaditya, 2025-05-19:
Thanks for the code snippet @etaoxing although this doesn't seem to solve the problem. I think there is something wrong with the underlying implementation and Featherstone seems to produce incorrect gradients even when a ball is attached to a prismatic joint.

I'm not too sure how to fix that. Please let me know if you ran into anything similar when you worked on rewarped

JonathanKuelz, 2025-05-19:
Unfortunately, I can't contribute a solution, but this matches my experience of having problems with autodifferentiating through the Featherstone integrator, as reported in #650  and #655. While I mostly encountered problems when computing gradients with respect to inertial parameters, your information that gradients seem to (initially) be more reasonable when setting `update_mass_matrix_every=1` could indicate that there is a shared common issue somewhere in how inertial properties of the bodies are computed/used.