NVIDIA/warp #740: [DOCS] Reorganize the Documentation

christophercrouzet, 2025-05-19:
## Category

- [ ] Report an error in the documentation.
- [ ] Request for something to be documented.
- [x] Suggestion to improve the documentation.
- [ ] Other (please explain)

## Description

- Change the Python reference page to strictly be a listing of the API, and move the remaining content into a dedicated user guide page, aligning it with our existing notebooks.
- Rename the “Built-Ins Reference” page to “Language Reference”.
- Ensure both reference pages have a clear table of contents on their landing pages that lists and groups all API components in a compact format. The detailed content should be stored on separate pages dedicated to each group (e.g.: Arrays, Streams, Graphs, Spatial Math, Time Primitives, Geometry, ...).
- Remove the simulation reference section as it's being moved to Newton, and rename the “Core Reference” page to “API Reference”.

shi-eric, 2025-07-18:
I guess you should create several sub-issues. At least the task about removing the simulation reference can't be done until November.

shi-eric, 2025-12-29:
Addressed by 34e132073b49c805c24726da62ca26202be59fd2