NVIDIA/warp #741: [BUG] Failed to run fem examples

FantasyVR, 2025-05-20:
### Bug Description

```
(lwarp) PS F:\warp> python -m warp.examples.fem.example_mixed_elasticity
Warp 1.7.1 initialized:
   CUDA Toolkit 12.8, Driver 12.8
   Devices:
     "cpu"      : "Intel64 Family 6 Model 165 Stepping 5, GenuineIntel"
     "cuda:0"   : "NVIDIA GeForce RTX 3060" (12 GiB, sm_86, mempool enabled)
   Kernel cache:
     C:\Users\Peng\AppData\Local\NVIDIA\warp\Cache\1.7.1
Module warp.fem.space.restriction.dyn.fill_element_node_indices_Grid2D_BoundarySides_Grid2D_Square_S2__69cf40fd 6eedfc7 load on device 'cuda:0' took 23.33 ms  (error)
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "F:\warp\warp\examples\fem\example_mixed_elasticity.py", line 289, in <module>
    example.step()
  File "F:\warp\warp\examples\fem\example_mixed_elasticity.py", line 189, in step
    u_bd_test = fem.make_test(space=self._u_space, domain=boundary)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "F:\warp\warp\fem\field\__init__.py", line 74, in make_test
    space_restriction = make_space_restriction(
                        ^^^^^^^^^^^^^^^^^^^^^^^
  File "F:\warp\warp\fem\space\__init__.py", line 89, in make_space_restriction
  File "F:\warp\warp\fem\space\__init__.py", line 89, in make_space_restriction
  File "F:\warp\warp\fem\space\__init__.py", line 89, in make_space_restriction
    return SpaceRestriction(
  File "F:\warp\warp\fem\space\__init__.py", line 89, in make_space_restriction
  File "F:\warp\warp\fem\space\__init__.py", line 89, in make_space_restriction
    return SpaceRestriction(
           ^^^^^^^^^^^^^^^^^
  File "F:\warp\warp\fem\space\restriction.py", line 49, in __init__
    self._compute_node_element_indices(device=device, temporary_store=temporary_store)
  File "F:\warp\warp\fem\space\__init__.py", line 89, in make_space_restriction
    return SpaceRestriction(
           ^^^^^^^^^^^^^^^^^
  File "F:\warp\warp\fem\space\__init__.py", line 89, in make_space_restriction
  File "F:\warp\warp\fem\space\__init__.py", line 89, in make_space_restriction
  File "F:\warp\warp\fem\space\__init__.py", line 89, in make_space_restriction
    return SpaceRestriction(
           ^^^^^^^^^^^^^^^^^
  File "F:\warp\warp\fem\space\restriction.py", line 49, in __init__
    self._compute_node_element_indices(device=device, temporary_store=temporary_store)
  File "F:\warp\warp\fem\space\restriction.py", line 83, in _compute_node_element_indices
    wp.launch(
  File "F:\warp\warp\context.py", line 5853, in launch
  File "F:\warp\warp\context.py", line 5853, in launch
    module_exec = kernel.module.load(device, block_dim)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "F:\warp\warp\context.py", line 2321, in load
    raise (e)
  File "F:\warp\warp\context.py", line 2297, in load
    with open(source_code_path, "w") as cu_file:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\Peng\\AppData\\Local\\NVIDIA\\warp\\Cache\\1.7.1\\wp_warp.fem.space.restriction.dyn.fill_element_node_indices_Grid2D_BoundarySides_Grid2D_Square_S2__69cf40fd_6eedfc7_p29372\\wp_warp.fem.space.restriction.dyn.fill_element_node_indices_Grid2D_BoundarySides_Grid2D_Square_S2__69cf40fd_6eedfc7.cu'
```

FantasyVR, 2025-05-20:
It seems that Windows couldn't handle too long file path (<=260 characters).

After setting the cache dir with command '$env:WARP_CACHE_PATH="F:\x", the example could run without errors poping up.

shi-eric, 2025-05-20:
Yeah, you might also be able to enable support for long file paths e.g. as described in #543

closed 2025-05-27 by christophercrouzet