NVIDIA/warp #759: [BUG] XPBD Integrator zero gradients in examples

Lancial, 2025-05-28:
### Bug Description

Target variable has no gradient in cloth_throw and softbody_properties examples when using XPBDIntegrator, so the optimization has no effect. SemiImplicitEuler integrator produces proper gradients. What is the cause of this issue here? Thanks!

### System Information

Warp version: 1.7.1, python 3.11

HussainAther, 2025-05-31:
Thanks for flagging this....really interesting behavior. I’ve also noticed XPBD integrators sometimes struggle with gradient flow in differentiable settings, especially if constraints introduce non-smooth behavior.

It might be helpful to check if the gradient is blocked due to how the constraints are projected, or if the loss function's dependency on simulation states is too weak. Curious if this persists across varying timestep sizes or stiffness parameters?

Happy to dig deeper or test a minimal example if that would be helpful.

shi-eric, 2025-11-06:
Sorry we weren't able to address this issue before removing `warp.sim` from the codebase. As of Warp v1.10, the `warp.sim` module has been removed. It was deprecated back in v1.8 and has been superseded by the [Newton physics engine](https://github.com/newton-physics/newton), which is now maintained as an independent Linux Foundation project with a redesigned API focused on robotics and robot learning.

If this issue is still relevant to your work, we'd encourage you to check out Newton. There's a [migration guide](https://newton-physics.github.io/newton/migration.html) available that should help with the transition. You can open issues in the [Newton repository](https://github.com/newton-physics/newton/issues) or ask questions in the [Newton Discussions](https://github.com/newton-physics/newton/discussions) section. Many of us from the Warp team are also involved with Newton and happy to help there.

closed 2025-11-06 by shi-eric