NVIDIA/warp #765: Fix Performance Regression For Kernel Launches

christophercrouzet, 2025-05-30:
As seen in the attached screenshot, the commit [a7084d8](https://github.com/NVIDIA/warp/commit/a7084d82) that improved support for tuples in kernels ([GH-506](https://github.com/NVIDIA/warp/issues/506)), caused a performance regression of ~12 μs caused by the changes to `warp.types.types_equal()`.

[image]

HussainAther, 2025-05-31:
Thanks for flagging this, @christophercrouze, good catch. That’s a nontrivial regression for applications making many small kernel calls.

Would a potential fix involve optimizing or caching inside types_equal() for commonly used tuples or types that are known to match? Alternatively, would memoizing type comparisons within a kernel launch context be worth exploring?

shi-eric, 2025-05-31:
You can take a look at @christophercrouzet's fix here: 83a43a6da59a9070b853aede42d14e2be5e0b504