NVIDIA/warp #769: [BUG] Warp Sim examples illegal hardware instruction on Mac Arm

fiona-yuxin-liu, 2025-06-02:
### Bug Description

Running `python example_bounce.py` on M1 MacBook Pro 
Getting

> Warp 1.7.1 initialized:
   CUDA not enabled in this build
   Devices:
     "cpu"      : "arm"
   Kernel cache:
     /Users/liuyuxin/Library/Caches/warp/1.7.1
Module warp.sim.collide 7905927 load on device 'cpu' took 0.68 ms  (cached)
zsh: illegal hardware instruction  python example_bounce.py

This issue appears to occur in all examples that use `warp.sim`.

yzx9, 2025-06-03:
Maybe related: #552

shi-eric, 2025-08-12:
@fiona-yuxin-liu: @c0d1f1ed developed a fix for this and it's now merged into the main branch. This fix should be in Warp v1.9.0, but if you want to try it now, you can build Warp from source on your local machine.

closed 2025-09-25 by shi-eric