NVIDIA/warp #781: [REQ] Rapids Memory Manager (RMM) Support

shi-eric, 2025-06-10:
### Description

https://github.com/rapidsai/rmm

Note that Warp does not support any external memory allocators, which makes this task less trivial.

### Context

Improve interop with PyTorch

coreyjadams, 2025-06-10:
Thank you for opening this issue!

For posterity: If/when this feature is implemented, the physicsnemo package has several model pipelines that combine pytorch + rapids + warp that could take advantage of this support.  In particular, the DoMINO datapipeline + model has warp kernels in both preprocessing and model implementation in pytorch.  Preprocessing also uses rapids' cuml package - and combining all of these bumps up against memory limits since pytorch will allocate most of the memory even when it's not in use.

I'm happy to be a tester if/when you need one.

shi-eric, 2026-04-18:
Interface added in b7ced88be290f18118865a647f73f5cc6c6728f4

Future work should support memory tracking like #1269 did for native allocators