NVIDIA/warp #789: [REQ] Type punning for Warp builtin types

phil-holiday, 2025-06-16:
### Description

It would be convenient if we could do type punning for Warp types, so we can convert between different primitives (ex. between float32 and int32) or possibly even structs. In CUDA C++, you can achieve this either with unions, `reinterpret_cast`, or `memcpy` (or `cuda::std::bit_cast` in recent versions if you like modern C++). But Warp cannot do any of these yet.

### Context

We're trying to find the minimum value (float32) inside an array, along with its index (int32). The most straightforward way to do this (without additional memory allocations) is to use `wp.tile_argmin()` inside the tile block and then use atomic CAS to globally reduce the minimum values. I've noticed that the `wp.atomic_cas()` function was recently added in the main branch, but there isn't a way to convert a (float32, int32) tuple into a int64 to utilize 64-bit CAS operations.

I think supporting this in Warp should be straightforward since CUDA already supports it - although with the caveat that it won't work with autodiff (which I'm perfectly fine with).

mmacklin, 2025-06-30:
Thanks Phil, sounds like a useful feature to me. You may be able to unblock yourself for now by using a native snippet, i.e.:

https://nvidia.github.io/warp/modules/differentiability.html#custom-native-functions

That takes the two values and packs them into an int64.

In the future we may look to provide a mechanism to cast values directly, e.g.: `wp.cast()`.

shi-eric, 2025-10-17:
@christophercrouzet added `wp.cast()` in 0b2a550628814bdf4932eeb29586b82b9cf08c1f, which will be part of the v1.10 Warp release.