NVIDIA/warp #809: Address libmathdx-related compile time regressions introduced with libmathdx 0.2.1

shi-eric, 2025-06-27:
Performance testing revealed some regressions in the compile times with Warp 1.8 resulting from the libmathdx upgrade, for example:

```
| Change   | Before [a447d70c] <v1.7.2^0>   | After [e74e05be] <release-1.8~3>   |   Ratio | Benchmark (Parameter)                                          |
|----------|--------------------------------|------------------------------------|---------|----------------------------------------------------------------|
| +        | 2.29±0s                        | 4.48±0.01s                         |    1.96 | tile.compile_cholesky.ColdCompileCholeskyLTO.time_cuda_codegen |
| +        | 1.29±0s                        | 1.73±0s                            |    1.35 | tile.compile_cholesky.WarmCompileCholeskyLTO.time_cuda_codegen |
| +        | 7.86±0.02s                     | 11.8±0.08s                         |    1.5  | tile.compile_fft.ColdCompileFftLTO.time_cuda_codegen           |
| +        | 2.02±0s                        | 2.91±0s                            |    1.44 | tile.compile_fft.WarmCompileFftLTO.time_cuda_codegen           |
| -        | 13.8±0.02s                     | 12.2±0.01s                         |    0.88 | tile.compile_gemm.ColdCompileGemmLTO.time_cuda_codegen         |
```

daedalus5, 2025-07-07:
Closed with [664e695](https://github.com/NVIDIA/warp/commit/664e695bfcb5ea9ec8f9fd40a5ca650c303a2787)

shi-eric, 2025-07-14:
@daedalus5 Seems like we don't have a changelog entry for this. Can you add one? Since the libmathdx version bump is just a perf bugfix as far as Warp is concerned, we can include it in the 1.8.1 release.