NVIDIA/warp #865: [REQ] External source support

nvlukasz, 2025-07-23:
### Description
Support using external C++/CUDA source in Warp kernels. This allows leveraging existing code without rewriting or redefining everything in Warp at the Python level. We should support custom native types and builtin functions that can work with these types.

### Context
Autodesk Neon/XLB integration relies on this feature.

massimim, 2026-10-01:
Extending @nvlukasz 's description.

Add support for implementing Warp-binding mechanisms for C++/CUDA types and functions through the following operations:
- [ ] Runtime registration of a C++/CUDA function
- [ ] Runtime registration of a C++/CUDA type
- [ ] Definition of user-defined namespaces in Warp to group external types and functions
- [ ] Listing of user-defined C++/CUDA libraries that accompany the external types/functions
- [ ] Configuration of NVRTC to parse (parsing only, not compiling) types and functions from the C++ standard library. It is common to Warp-bind CUDA functions/types that are defined in files where types from the standard library are included. The goal is to be able to process any file from a C++/CUDA project compiled with nvcc for Warp bindings.
- [ ] Object-oriented syntax for external types (optional)
- [ ] Warp-binding support for CPU-only target machines

shi-eric, 2026-10-07:
@daedalus5 can you please create appropriate sub-issues from @massimim's comment and start assigning them to releases?