NVIDIA/warp #922: [BUG] Inconsistent sign + inaccuracy in SDF computation

ehsanhaghighat, 2025-08-19:
### Bug Description

Hi Team, 

We have a pipeline that relies on accurate SDF inputs, previously computed using Open3D. Before choosing Open3D, I extensively tested all available open-source options and I found Open3D implementation as the most robust (to mesh issues) and most accurate implementations. I tested on sampled data from ABC datasets. 

I recently tested warp for such a computation. I did observe a significant speed up over open3d, but I also found inconsistent results. I tested `mesh_query_point`, `mesh_query_point_sign_normal` and `mesh_query_point_sign_winding_number`. Here are a few images of SDF evaluated on random geometries: 

Random sign change: 

[image]
vs 

[image]

Unexpected artifacts: 

[image]

Not recognizing inside and outside correctly:

[image]

One last example: 

[image]

It would be great to add some unit tests based on Open3D outputs.

shi-eric, 2025-08-21:
Thank you @ehsanhaghighat for the detailed bug report. @AnkaChan will look into this issue. It might help to provide links to the assets you used for the figures.

AnkaChan, 2025-08-28:
Hi [ehsanhaghighat](https://github.com/ehsanhaghighat), thank you for reporting this. 
However, from you figure I cannot easily tell what wrong. Can you highlight the parts that is wrong? 
Can you be more descriptive about the errors and artifacts?

sai-ganesh-c-infinity, 2026-04-03:
Hi @AnkaChan @shi-eric,

I tested building the sdf using the new function `wp.mesh_query_point_sign_parity()` and the results look promising. The values are comparable to Open3D in the order of 1e-4 and the signs are consistent as well. So I think this is good to close.

Inaccuracy fixed:
[image]

Sign corrected:
[image]