NVIDIA/warp #926: [BUG] Local matrix assignment with `a[i][j] = b` fails silently, must use `a[i, j] = b` instead

j3soon, 2025-08-20:
### Bug Description

Warp function/kernel in the following form does not correctly perform matrix element assignment:

```sh
# A simple function/kernel to mimic the behavior of `A @ B`.
@wp.func
def matmul2x2(A: wp.mat22, B: wp.mat22) -> wp.mat22:
    C = wp.mat22()
    for i in range(2):
        for j in range(2):
            C[i][j] = A[i][0] * B[0][j] + A[i][1] * B[1][j]  # does not change C
    return C
```

To get the correct result, `C[i, j] = ...` must be used instead. I suspect this may be caused by `C[i][j]` first copies a matrix slice `C[i]`, and then the assignment is performed on that copy. This issue doesn't seem to happen on arrays.

I believe the correct behavior is either make `C[i][j] = ...` to correctly perform assignment, or at least the debugging message should report some warning during kernel compilation.

<hr>

To reproduce:

```py
import numpy as np
import warp as wp

# Enable debugging mode
wp.config.lineinfo = True
wp.config.print_launches = True
wp.config.verbose = True
wp.config.verbose_warnings = True
wp.config.mode = "debug"
wp.set_module_options({"mode": "debug"})
wp.config.cache_kernels = True
wp.config.verify_cuda = True
wp.config.verify_fp = True

wp.clear_kernel_cache()

wp.init()  # initialize Warp

# The incorrect behavior will still happen when combining them into a single kernel without the use of `@wp.func`.

@wp.func
def matmul2x2(A: wp.mat22, B: wp.mat22, correct: bool) -> wp.mat22:
    C = wp.mat22()
    for i in range(2):
        for j in range(2):
            if correct:
                C[i, j] = A[i][0] * B[0][j] + A[i][1] * B[1][j]
            else:
                C[i][j] = A[i][0] * B[0][j] + A[i][1] * B[1][j]
    return C

@wp.kernel
def test(a: wp.mat22,
         b: wp.mat22,
         c: wp.array2d(dtype=wp.float32),
         correct: bool):
    ret = matmul2x2(a, b, correct)
    for i in range(2):
        for j in range(2):
            c[i][j] = ret[i][j]
            
n = 2
device = "cuda"
a_host = np.array(
    [[1, 2],
     [3, 4]],
    dtype=np.float32)
b_host = np.array(
    [[5, 6],
     [7, 8]],
    dtype=np.float32)

a = wp.mat22(a_host)
b = wp.mat22(b_host)
c = wp.empty((n, n), dtype=wp.float32, device=device)
d = wp.empty((n, n), dtype=wp.float32, device=device)

# Launch kernel with 2D grid (2x2 threads)
wp.launch(kernel=test, dim=1, inputs=[a, b, c, False], device=device)
wp.launch(kernel=test, dim=1, inputs=[a, b, d, True], device=device)

print("answer =\n", a_host @ b_host)
print("a x b (incorrect) =\n", c.numpy())
print("a x b (correct) =\n", d.numpy())
```

and then run it with `python3 test.py`, and observe the following:

```
# python3 test.py 
Warp 1.8.1 initialized:
   CUDA Toolkit 11.8, Driver 12.4
   Devices:
     "cpu"      : "x86_64"
     "cuda:0"   : "NVIDIA L40" (44 GiB, sm_89, mempool enabled)
   Kernel cache:
     /root/.cache/warp/1.8.1
kernel: test dim: 1 inputs: [<warp.types.mat22f object at 0x147cb8eb7840>, <warp.types.mat22f object at 0x147cb8eb7740>, array(shape=(2, 2), dtype=float32), False] outputs: [] device: cuda:0
Module __main__ 08a4c15 load on device 'cuda:0' ...
    Compile CUDA ...
    Compile CUDA took 519.84 ms
Module __main__ 08a4c15 load on device 'cuda:0' took 620.18 ms  (compiled)
kernel: test dim: 1 inputs: [<warp.types.mat22f object at 0x147cb8eb7840>, <warp.types.mat22f object at 0x147cb8eb7740>, array(shape=(2, 2), dtype=float32), True] outputs: [] device: cuda:0
answer =
 [[19. 22.]
 [43. 50.]]
a x b (incorrect) =
 [[0. 0.]
 [0. 0.]]
a x b (correct) =
 [[19. 22.]
 [43. 50.]]
```

### System Information

- Python 3.10.13
- Warp 1.8.1

(Can also be reproduced on Google Colab with Python 3.12.11)

c0d1f1ed, 2025-08-20:
This seems like a duplicate of #793, which was recently fixed. Please give [v1.9.0rc1](https://github.com/NVIDIA/warp/releases/tag/untagged-89f9c890f90fe79a88a5) a try.

j3soon, 2025-09-02:
> This seems like a duplicate of [#793](https://github.com/NVIDIA/warp/issues/793), which was recently fixed. Please give [v1.9.0rc1](https://github.com/NVIDIA/warp/releases/tag/untagged-89f9c890f90fe79a88a5) a try.

Confirmed the fix on current `main`, thanks! Closing this issue.

closed 2025-09-02 by j3soon