NVIDIA/warp #930: [QUESTION] Incomplete Fix: Commit 0029c21 "Remove Broadcasting Operations on `vec`/`mat` Types"

j3soon, 2025-08-24:
During a rebase of PR https://github.com/NVIDIA/warp/pull/887, I observed that commit 0029c21f527d33abf1c60d5517a74cbf1aad0193 only partially removes the broadcasting operations, the followings are code snippets that I believe should be removed but may be overlooked in that commit.

Search `input_types={"a": vector(length=Any, dtype=Scalar), "b": Scalar}`:

https://github.com/NVIDIA/warp/blob/d10c7d4f00508008711f5d30e9bbde6fad7dffca/warp/builtins.py#L7703-L7730

https://github.com/NVIDIA/warp/blob/d10c7d4f00508008711f5d30e9bbde6fad7dffca/warp/builtins.py#L7738-L7751

https://github.com/NVIDIA/warp/blob/d10c7d4f00508008711f5d30e9bbde6fad7dffca/warp/builtins.py#L7891-L7932

Search `Type a, vec_t<Length, Type> b`:

https://github.com/NVIDIA/warp/blob/d10c7d4f00508008711f5d30e9bbde6fad7dffca/warp/native/vec.h#L346-L355

https://github.com/NVIDIA/warp/blob/d10c7d4f00508008711f5d30e9bbde6fad7dffca/warp/native/vec.h#L381-L391

https://github.com/NVIDIA/warp/blob/d10c7d4f00508008711f5d30e9bbde6fad7dffca/warp/native/vec.h#L1306-L1319

https://github.com/NVIDIA/warp/blob/d10c7d4f00508008711f5d30e9bbde6fad7dffca/warp/native/vec.h#L1348-L1361

In addition to the deletion of the lines above, if some tests have been relying on auto-broadcasting, should also modify those tests to switch to manual broadcasting.

christophercrouzet, 2025-08-26:
Thank you @j3soon!

The intention for the commit [0029c21](https://github.com/NVIDIA/warp/commit/0029c21f527d33abf1c60d5517a74cbf1aad0193) is to revert the broadcasting operators there were introduced in commit [e6c5012](https://github.com/NVIDIA/warp/commit/e6c50124498d8546dff29f427439ff1daf37f3bb), that is only `add`/`sub`. The `mul`/`div` operators were added much earlier, and are necessary for uniform scaling, so we want to keep them. Apologies for the confusion!

That being said, as you pointed out, there's indeed some leftover definitions in the `vec.h` file needs to be cleaned up! We've open a MR internally to do just thank, thanks again!

christophercrouzet, 2025-08-26:
This has now been merged in commit [cc608ce](https://github.com/NVIDIA/warp/commit/cc608ceae198cd13687d2aec587040b2046e87ab).

closed 2025-08-26 by christophercrouzet