NVIDIA/warp #940: [BUG] 5070ti Warp CUDA error

xxll88, 2025-08-30:
### Bug Description

Warp CUDA error: Failed to get driver entry point 'cuDeviceGetUuid' (CUDA error 1)
Warp CUDA error: Function cuDeviceGetUuid_f: a suitable driver entry point was not found
Warp CUDA error 36: API call is not supported in the installed CUDA driver (in function cuda_init, C:\g\163821891\warp\native\warp.cu:277)

flrs, 2025-10-04:
Same issue as in [IsaacSim/issues/229](https://github.com/isaac-sim/IsaacSim/issues/229), with a possible temporary solution of downgrading GPU drivers/CUDA

shi-eric, 2025-10-04:
> Same issue as in [IsaacSim/issues/229](https://github.com/isaac-sim/IsaacSim/issues/229), with a possible temporary solution of downgrading GPU drivers/CUDA

I don't think it's the **same** issue since the user in https://github.com/isaac-sim/IsaacSim/issues/229 was using Warp v1.7.1, which from my first comment has a bug that would prevent it from working with a R580 driver. The issue in this thread concerns users running into a similar issue despite using Warp v1.9.0.

Zvch17, 2025-10-30:
I met the same problem. 
$ python -m warp.tests.test_devices
Warp 1.9.1 initialized:
   CUDA Toolkit 12.8, Driver 13.0
   Devices:
     "cpu"      : "x86_64"
     "cuda:0"   : "NVIDIA GeForce RTX 5090" (31 GiB, sm_120, mempool enabled)
   Kernel cache:
     /home/admin1/.cache/warp/1.9.1
test_devices_can_access_self_cpu (__main__.TestDevices.test_devices_can_access_self_cpu) ... ok
test_devices_can_access_self_cuda_0 (__main__.TestDevices.test_devices_can_access_self_cuda_0) ... ok
test_devices_get_cuda_device_functions_cuda_0 (__main__.TestDevices.test_devices_get_cuda_device_functions_cuda_0) ... ok
test_devices_map_cuda_device_cuda_0 (__main__.TestDevices.test_devices_map_cuda_device_cuda_0) ... ok
test_devices_sm_count_cpu (__main__.TestDevices.test_devices_sm_count_cpu) ... ok
test_devices_sm_count_cuda_0 (__main__.TestDevices.test_devices_sm_count_cuda_0) ... ok
test_devices_unmap_imaginary_device (__main__.TestDevices.test_devices_unmap_imaginary_device) ... ok
test_devices_verify_cuda_device_cpu (__main__.TestDevices.test_devices_verify_cuda_device_cpu) ... ok
test_devices_verify_cuda_device_cuda_0 (__main__.TestDevices.test_devices_verify_cuda_device_cuda_0) ... ok

----------------------------------------------------------------------
Ran 9 tests in 0.071s

OK

$ nvidia-smi
Thu Oct 30 18:49:26 2025       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 580.95.05              Driver Version: 580.95.05      CUDA Version: 13.0     |
+-----------------------------------------+------------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 5090        Off |   00000000:01:00.0  On |                  N/A |
|  0%   37C    P8             21W /  575W |     418MiB /  32607MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+

+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI              PID   Type   Process name                        GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A            1410      G   /usr/lib/xorg/Xorg                      176MiB |
|    0   N/A  N/A            2199      G   /usr/bin/gnome-shell                     35MiB |
|    0   N/A  N/A          338623      G   /usr/share/code/code                     63MiB |
|    0   N/A  N/A          383241      G   ...ersion=20251029-170052.874000         92MiB |
+-----------------------------------------------------------------------------------------+

my nvidia driver is nvidia-driver-580-open/jammy-updates,jammy-security,now 580.95.05-0ubuntu0.22.04.1 amd64

shi-eric, 2026-03-18:
Haven't heard anything on this issue for a while, so I'm going to close it. Please reopen if you find that you're still have issues with recent Warp versions.