NVIDIA/warp #941: Issue with tile assignment involving composite types

shi-eric, 2025-09-01:
> Potentially related:
> 
> ```py
> import numpy as np
> import warp as wp
> 
> vec256f = wp.types.vector(length=256, dtype=wp.float32)
> 
> @wp.kernel
> def test(a: wp.array2d(dtype=wp.float32)):
>     i = wp.tid()
>     R = wp.tile_zeros(shape=(1,), dtype=vec256f)
>     R[0][i] = wp.float32(i)
>     a[i // 16, i % 16] = R[0][i]  # incorrect
>     # a[i // 16, i % 16] = wp.float32(i)  # correct
> 
> TILE_SIZE = 256
> device="cuda"
> a = wp.zeros((16, 16), dtype=wp.float32, device=device)
> wp.launch(kernel=test, dim=TILE_SIZE, outputs=[a], block_dim=TILE_SIZE, device=device)
> wp.synchronize()
> print(a.numpy())
> ```
> 
> ```
> kernel: test dim: 256 inputs: [] outputs: [array(shape=(16, 16), dtype=float32)] device: cuda:0
> [[0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]
>  [0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]
>  [0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]
>  [0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]
>  [0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]
>  [0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]
>  [0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]
>  [0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]
>  [0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]
>  [0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]
>  [0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]
>  [0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]
>  [0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]
>  [0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]
>  [0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]
>  [0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0. 0.]]
> ```
> 
> I'll try to test this on v1.9.0 later. 

 _Originally posted by @j3soon in [#926](https://github.com/NVIDIA/warp/issues/926#issuecomment-3217464865)_

j3soon, 2025-09-02:
May be related to this snippet removed in commit 195f7743cac44e0a7872a7ac78dd44a0f16d21ec:

https://github.com/NVIDIA/warp/blob/f05553aec118744081963b7f8291c848a748013b/warp/codegen.py#L2464-L2515

shi-eric, 2025-10-17:
Support added by @daedalus5 in 36bbae03111a928f5c9783ebe008b0515d8bd4d5

closed 2025-10-17 by shi-eric